beagleboard/kernel #102: CPU Frequency Scaling doesn't work

ViktorButskikh, 2016-03-07:
Hello.
Recently I decided update rootfs on my BBB from buildroot-2014.02 to 2016.02.
As a part of update I have migrated from 3.8 kernel to 4.4.3.
Kernel source code 4.4.3  were taken from https://github.com/beagleboard/linux.git. For build kernel I took default config bb.org. After successful build rootfs and boot my system I found that cpufreq doesnt't work:
` cpufreq-info 
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 0.00 ms.`

It's strange behavior for me, because on 3.8.13 kernel everything work fine.

May be it's happened because I haven't integrate bb.org overlay to my system, yet...
Could you help me found way for solve this?

RobertCNelson, 2016-03-07:
@ViktorButskikh cpufreq support (from v4.1.x branch) needs to be re-ported to v4.4.x, and as a side note, ti's decided to go a different way then the "voltdm" setup used in v4.1.x...

So yeah, right now, no cpufreq in v4.4.x...

imyller, 2016-08-09:
@RobertCNelson Any update on this?

Are there working alternatives to cpufreq with v4.4.x?

RobertCNelson, 2016-08-09:
@imyller cpufreq scaling works fine on the top of the v4.4.x tree:

```
debian@test-bbb-5:~$ uname -r
4.4.16-ti-rt-r38
debian@test-bbb-5:~$ cpufreq-info 
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  driver: cpufreq-dt
  CPUs which run at the same hardware frequency: 0
  CPUs which need to have their frequency coordinated by software: 0
  maximum transition latency: 300 us.
  hardware limits: 300 MHz - 1000 MHz
  available frequency steps: 300 MHz, 600 MHz, 720 MHz, 800 MHz, 1000 MHz
  available cpufreq governors: conservative, ondemand, userspace, powersave, performance
  current policy: frequency should be within 300 MHz and 1000 MHz.
                  The governor "ondemand" may decide which speed to use
                  within this range.
  current CPU frequency is 300 MHz.
  cpufreq stats: 300 MHz:99.63%, 600 MHz:0.06%, 720 MHz:0.05%, 800 MHz:0.08%, 1000 MHz:0.18%  (769)
```