beagleboard/kernel #103: mmcblk1 does not appear for new beaglebone black board

awadhesh39, 2019-11-21:
I have some new beagle bone black boards where I cannot find /dev/mmcblk1 only /dev/mmcblk0 is appearing for SD Card. I am using kernel version 3.8 and u-boot version v2013.07. Also I found at u-boot prompt command fatls mmc 1 shows invalid mmc or bad device.

After searching I found in recent board emmc driver is updated for that reason old emmc code of kernel and u-boot cannot initialise emmc and also found that in kernel version 4+ fixes are there.  

I want to know is there any patches or fixes in old kernel version 3.18 and u-boot v2013  so that I can fix this issue without replacing my whole kernel and u-boot code.

RobertCNelson, 2019-11-21:
@awadhesh39 this is a known issue on really old 3.8.13 based kernels..  Fixed on 3.8.13-bone80

https://gist.github.com/RobertCNelson/39faf80ddc9fcefae74dce2c6ca2eb45

Regards,

awadhesh39, 2019-11-25:
Thanks for your reply.. 

I have tried to run steps as provided above getting error:
....
.....
CC credential-cache.o
In file included from http.c:1:0:
http.h:6:23: fatal error: curl/curl.h: No such file or directory
 #include <curl/curl.h>
                       ^
compilation terminated.
make: *** [http.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from http-walker.c:4:0:
http.h:6:23: fatal error: curl/curl.h: No such file or directory
 #include <curl/curl.h>
                       ^
......
........................................................................
I have kernel version "3.8.13-00745-g6e0b32d-dirty" [uname -r] hence I replaced it by 

sed -i -e 's:3.8.13-bone86:3.8.13-00745-g6e0b32d-dirty:g' recipe.sh

I need some more clarifications:
1. Is this procedure applicable version kernel: 3.8.13-00745-g6e0b32d-dirty?
2. Where to run all these commands in kernel source folder?

RobertCNelson, 2019-11-25:
@awadhesh39:

curl/curl.h : install the package on your linux distribution that provides that..

3.8.13-00745-g6e0b32d-dirty = is not MY release, first go figure out where you built that from, then add the two patches manually..

I provide support for only MY releases, which the yakbuild script can build..

Regards,

RobertCNelson, 2019-11-25:
PS, this repo is deprecated:

https://github.com/beagleboard/kernel/blob/master/README.md

This repo is now deprecated
Please see https://github.com/beagleboard/linux for the latest official kernels.

awadhesh39, 2019-11-28:
Thanks for your reply...
It is working now for new eMMc.

closed 2019-11-28 by awadhesh39